paulscherrerinstitute/ecmccfg #24: low default current for motor drivers

kivel, 2019-12-03:
By default, some slaves are set to full current. This could lead to damage to the motor in case the desired configuration is not properly loaded.

By implementing a default configuration for such slaves - with a low current setting - the issue could be resolved.

Alternatively, a separate mechanism to instantiate such 'problematic' slaves could be implemented into ECMC, similar to 'ConfigOrDie'.

anderssandstrom, 2020-03-09:
@kivel 
Made a different solution (or part of it). Added a iocsh command, ecmcFileExist(), that check if a file exists. If the file does not exists, then EPICS execution can be halted.
Example:
_ecmcFileExist("${ECMC_CONFIG_ROOT}ecmc${HW_DESC}${CONFIG}.cmd",1)
${SCRIPTEXEC} ${ECMC_CONFIG_ROOT}ecmc${HW_DESC}${CONFIG}.cmd_

in icshwi/ecmccfg, ecmcFileExist() now checks/ensures that files exist (also db-files) in most scripts and commands, and otherwise ecmc/EPICS exits. 
So, if the ecmc ioc is running, all configuration files should be loaded.

Probably the low current settings also should be added to the configuration files but risk for failure is significantly lower now.

anderssandstrom, 2023-10-27:
@kivel, sure lets close.

closed 2023-10-27 by kivel