paulscherrerinstitute/ecmccfg #40: hardcoded EPICS record processing frequency

kivel, 2020-10-19:
I'm observing that _ecmccfg_ is disregarding "ECMC_SAMPLE_RATE_MS"

I think the issue is within "initAll.iocsh"
https://github.com/paulscherrerinstitute/ecmccfg/commit/b483b1f61225d4db058f450e03a706f0207dc663#r43369105

@anderssandstrom can you confirm the same at ESS?

anderssandstrom, 2020-10-20:
Not sure I understood the issue completely..  Anyway, for me it seems to work (tested 10ms and 100ms and checked some pv with camonitor).
You could describe the ECMC_SAMPLE_RATE_MS as a global variable you can change/set at any time in your startup script. All calls to dbLoadRecord/Tempalte executed after the changed ECMC_SAMPLE_RATE_MS will have records based on the new value.
In initAll.cmd the ECMC_SAMPLE_RATE_MS is initialised to a default value of 10ms.
Maybe would be nicer to have it as a parameter to all the "functions"?!

kivel, 2020-10-20:
Hi Anders

My bad, I didn't get the logic. I was declaring it globally before require. That didn't work of course 😊

All good now.

best

Niko

--
Paul Scherrer Institute
Dr. Niko Kivel
Forschungsstrasse 111
WBGB/017
5232 Villigen PSI
Switzerland

phone: +41 56 310 2226
email: niko.kivel@psi.ch

closed 2020-11-18 by kivel