paulscherrerinstitute/ecmccfg #158: potential duplicate MCU-errorID

kivel, 2022-03-11:
@anderssandstrom is the data in both PVs identical, even though stemming from different sources?
If the information in both PVs is actually identical, I prefer to remove it from the motorRecord template.

https://github.com/paulscherrerinstitute/ecmccfg/blob/1855d64b1a69ebdeaf299b67e1cb8f035e982b56/db/core/ecmcMotorRecord.template#L66

https://github.com/paulscherrerinstitute/ecmccfg/blob/1855d64b1a69ebdeaf299b67e1cb8f035e982b56/db/core/ecmcAxis.db#L488

kivel, 2022-03-29:
@anderssandstrom would you mind looking into this and advise?

anderssandstrom, 2022-03-30:
@kivel, in most cases they are the same but I think there could be situations when also motor record sets an  error code. Lets check with Torsten
@tboegi, are you in motor record relying on the error from the mcu or do you also sometimes set other errors?

tboegi, 2022-03-30:
No, we just forward the ErrorId.
And that is not the motorRecord (as record), but the model 3 driver,
which allows a longin record.
So having the very same record-definition twice does not do harm, AFAIK

kivel, 2022-03-31:
@tboegi thanks for the info.
It's not about harm, it's about duplication.
If there are no objections, I'm going to remove the PV in the motorRecord-specific template.

commit 440ed4e0 (2022-04-01): removed `record(longin, "$(PREFIX)$(MOTOR_NAME)-Err")` from `ecmcMotorRecord.template`
closed #158
closed 2022-04-01 by kivel