paulscherrerinstitute/ecmccfg #166: PV-property naming to long

anderssandstrom, 2022-03-30:
Some PVs do not comply with the ESS naming policy allowing 25chars for "property" part.
The following PVs have been identified to be too long when using the ecmcESSnaming macro:
* EL1252 timestamp PVs: #m0s101-latchTime01-fallingTS
* Incremental encoder latch PV: #m0s182-Enc01-LtchAutRstSp
* EL5101-0011: #m0s179-Enc01-PosAct-Array
* EL7037_POSIF: #m0s205-Pos01-StartType-RB
* Steppers: #m0s197-Drv01-SyncErr-Alrm
* DC drive typo: #m0s201--Drv01-SyncErr-Alrm
* Axis10-AllwCmdFrmPLCCmd-RB: PV property to long

I would like to shorten the names of these PVs
@kivel , What do you think?

kivel, 2022-03-30:
EL1252 timestamp PVs: #m0s101-latchTime01-fallingTS
--> m0s101-latch01-fallingTS (24 char)
Honestly, for the EL1252 they should not be like this in the first place.
https://github.com/paulscherrerinstitute/ecmccfg/blob/df58610bb897595969a63b71404bce9b83b46024/db/Beckhoff_1XXX/ecmcEL1252.substitutions#L8-L15
The `latchTime` comes only up in EP1258
https://github.com/paulscherrerinstitute/ecmccfg/blob/df58610bb897595969a63b71404bce9b83b46024/db/Beckhoff_1XXX/ecmcEP1258.substitutions#L17-L20

anderssandstrom, 2022-03-30:
Ahh, you're right. I just figured it was EL1252 without really checking. So lets harmonise so EL1252 and EP1258 looks similar.
Your suggestions m0s101-latch01-fallingTS and m0s182-Enc01-LchAutRstSp is fine with me.

kivel, 2022-03-30:
EL5101-0011: #m0s179-Enc01-PosAct-Array
--> EL5101-0011: #m0s179-Enc01-PosAct-Arr
This will however make this one different from the other `Array` PVs for the oversampled terminals.
Should we make them all the same? So a general `Array` --> `Arr`?

kivel, 2022-03-30:
Steppers: #m0s197-Drv01-SyncErr-Alrm
DC drive typo: #m0s201--Drv01-SyncErr-Alrm
==> `Alrm` --> `Alm` (Alm is the second typical abbreviation for Alarm)
Similar to the `Array` I presume _all_ `Alrm` should be changed to `Alm`

anderssandstrom, 2022-03-30:
Yes, I think that changing of "array" to "arr" is the best since the rest of the property is hard to change (#m0s179-Enc01-PosAct).
Changing all oversampled to "Arr" would lead to some frustration for sure but must be the right way forward..

kivel, 2022-03-30:
> Changing all oversampled to "Arr" would lead to some frustration for sure but must be the right way forward.

Yeah, sure, but this goes into v8, so breaks are to be expected. This is to be blamed on the person that insists on <=24 char for the property.

kivel, 2022-03-30:
> Regarding #m0s197-Drv01-SyncErr-Alrm we have already have several slaves with the following syntax:
> #m0s197-Drv01-SyncErrAlrm
> So that is probably an easier fix, right?

I'd like to make the names homogeneous. So if we have to shorten one PV to `Alm` _all_ will be changed. This make it more predictable. Otherwise, we will have to face questions why one ist `Alrm` and the other is `Alm`.

anderssandstrom, 2022-03-30:
We already have SyncErr-Alrm and SyncErrAlrm.
So standardising on SyncErrAlrm would solve this issue.
I think we have no "Alm" but 662 "Alrm" so I would prefer to not go for "Alm"

kivel, 2022-03-30:
> So standardising on SyncErrAlrm would solve this issue.

`SyncErrAlrm` it is

anderssandstrom, 2022-03-30:
Noticed now that we have some more inconsistencies:
* Wrn-Alrm and WrnAlrm
* Err-Alrm and ErrAlrm
* Stl-Alrm and StlAlrm
* Lim-Alrm
* Flw-Alrm
* QStp-Alrm
* KBusErr-Alrm
* ...

But most records are actually without the "-"-

I will also harmonise these by removing "-"

anderssandstrom, 2022-03-30:
Actually quite useful to look through this:
https://github.com/anderssandstrom/ecmccfg_ess_validator/blob/master/pvs.log

anderssandstrom, 2022-04-08:
### EL 1252:
```
ECMC_TEST_PREFIX:#m0s105-BI01-TimeRise
ECMC_TEST_PREFIX:#m0s105-BI01-TimeFall
ECMC_TEST_PREFIX:#m0s105-BI02-TimeRise
ECMC_TEST_PREFIX:#m0s105-BI02-TimeFall
ECMC_TEST_PREFIX:#m0s105-BI01-TimeRiseTS
ECMC_TEST_PREFIX:#m0s105-BI01-TimeFallTS
ECMC_TEST_PREFIX:#m0s105-BI02-TimeRiseTS
ECMC_TEST_PREFIX:#m0s105-BI02-TimeFallTS
ECMC_TEST_PREFIX:#m0s105-BI01-Stat
ECMC_TEST_PREFIX:#m0s105-BI02-Stat
ECMC_TEST_PREFIX:#m0s105-BI01
ECMC_TEST_PREFIX:#m0s105-BI02
```
### EL1258
```
ECMC_TEST_PREFIX:#m0s101-latchTime01-rising
ECMC_TEST_PREFIX:#m0s101-latchTime01-falling
ECMC_TEST_PREFIX:#m0s101-latchTime02-rising
ECMC_TEST_PREFIX:#m0s101-latchTime02-falling
ECMC_TEST_PREFIX:#m0s101-latchTime01-risingTS
ECMC_TEST_PREFIX:#m0s101-latchTime01-fallingTS
ECMC_TEST_PREFIX:#m0s101-latchTime02-risingTS
ECMC_TEST_PREFIX:#m0s101-latchTime02-fallingTS
ECMC_TEST_PREFIX:#m0s101-BI01-Stat
ECMC_TEST_PREFIX:#m0s101-BI02-Stat
ECMC_TEST_PREFIX:#m0s101-BI01
ECMC_TEST_PREFIX:#m0s101-BI02
ECMC_TEST_PREFIX:#m0s101-BI03
ECMC_TEST_PREFIX:#m0s101-BI04
ECMC_TEST_PREFIX:#m0s101-BI05
ECMC_TEST_PREFIX:#m0s101-BI06
ECMC_TEST_PREFIX:#m0s101-BI07
ECMC_TEST_PREFIX:#m0s101-BI08
```

I actually like the EL1252 syntax better, what do you think?

closed 2023-05-18 by anderssandstrom